necroware/nwX287 #2: Compatibility with the *B* and *C* variants

Jackowski3k, 2021-11-25:
Hello. 

Out there are multiple versions of the Dallas RTC chip, among them the DS12B887 and DS12C887
[image]

How does it looks with the compatibility? I have tried various alternatives I could find online and they didn't worked on my systems.

necroware, 2021-11-25:
You can use this module with different ICs, so it is heavily dependent on what you are using. I used it with BQ3285 and BQ4285 depending on the mainboard requirements and both variants worked perfectly. Due to the datasheets the DS12C885 (used in DS12C887) is an extended version of the DS12B885 (used in DS12B887), which is again an extended version of the DS1285. The extended versions are a drop-in replacement for the older versions accordingly. The BQ3285 should be identical with DS12C885 and I had no mainboards so far, where it didn't work.

steeviebops, 2023-07-12:
This board, as built, is functionally equivalent to the DS12B887. The main difference between the 12887 and 12B887 is that the B variant exposes pin 21 of the internal RTC. This is the "RAM clear" pin, used for clearing CMOS settings via a jumper. Pin 21 is absent from a standard DS12887 module so it can't be cleared using the jumper method.

The DS12C887 contains an extra "century byte" so I imagine that the RTC would need to be different. I've seen some references to a DS12C885 but it seems to be hard to find.

necroware, 2023-12-30:
Thank you very much. I mentioned this in the documentation. The PCB itself exposes pin 21 already, so it is only a question of which pins to cut. I guess this can be closed.

closed 2023-12-30 by necroware