necroware/nwX287 #7: Include the symbol cache

ndemarco, 2021-12-17:
I'm fairly new to KiCad. Actually, I just installed it. I do use Altium for work, though.

It seems the symbol cache is missing.
[image]

necroware, 2021-12-17:
Kicad generates the *-cache.lib file automatically. Please don't use KiCad 5.99, that is a development version and you will have many issues using that one. The current official version is 5.1.12

necroware, 2021-12-17:
By the way, if you want just to order the PCBs, you don't need to use KiCad at all. Simply go into releases and download the gerber files, which you can use to make the order.  https://github.com/necroware/nwX287/releases

ndemarco, 2021-12-17:
Thank you 2x today (not seeing ver 1.a, not seeing I installed KiCad
preview). I really must wake up! Your videos are great. Very well done,
precise and clear.

KiCad interests me, so I prefer to go that way.

necroware, 2023-12-30:
done

closed 2023-12-30 by necroware