necroware/nwX287 #10: "cmos checksum error defaults loaded" with BQ3285S RTC

theniwo, 2022-11-22:
Hi all.

I assembled an RTC-Module with the [nwX287](https://github.com/necroware/nwX287) PCB and a [BQ3285S](https://www.ebay.de/itm/324259367588?hash=item4b7f5d4aa4:g:jIkAAOSwwFZfNACP&amdata=enc%3AAQAHAAAAoFxNkcSkXAFhd86O3EtQnLejKJPnKylCDT7X8lu4aMR%2BZ3wcUTOh0KRdHmtVZdezL%2B34F16hWCoy8vceY7GQzMJU2DyV3ORKM6XW6Hy1gYfQs20hS6FVZi%2FEAiIMPzdylbkXU48CAUtEErS7nenf4Xx8al7mQvTNwnfAy4yeYs1s0GalKEVtTDRjPCaaefKjVfJfF5TjYaRJmghVcLOWxXc%3D%7Ctkp%3ABk9SR86w2OSSYQ) RTC.

Now I am getting the "cmos checksum error defaults loaded".

The Battery Error is not indicated and the battery is fresh and has voltage.

I testet

* Loading BIOS or Setup defaults

* Flashing the BIOS with the latest ([WI7C 040197](https://theretroweb.com/motherboard/bios/wi7c-5f5ab87c0cb5d152278531.exe)) Award from [theretroweb](https://theretroweb.com/motherboards/s/msi-ms-5129-tr5#downloads)

* Checked continuity on all pins and legs on the RTC Module

What I noticed was that the time drifted forward and the seconds where erroraneous.

It goes like this: 1 - 19, 40, 0

What is going on here? Did I use the wrong RTC Chip? 

May it be a counterfeit?

Did I damage it because it got a bit hot during soldering?

Can I rule the BIOS EEPROM out since it should be an issue with the RTC?

I have the original ODIN OEC12C887. Should I mod it to rule the BIOS EEPROM out?

necroware, 2022-12-05:
May be the BQ3285S is defective. The BIOS chip is not related.

necroware, 2023-12-30:
I don't think, that this is related to the module itself. Looks like an incompatibility or defect of the chip itself.

closed 2023-12-30 by necroware

theniwo, 2025-05-22:
I changed the IC and it works now. The one I had was probably defective, or a counterfeit

Front:
[image]

Back
[image]

Working IC
[image]