necroware/nwX287 #11: BQ3287MT

adamz464, 2022-11-26:
Unfortunately doesn't work (Alaris Cougar)...

necroware, 2022-12-05:
Ok, then I'd like to know, which module was previously in there, if the board works reliably with the original IC and what means "doesn't work"? Black screen, no boot, no POST codes?

adamz464, 2022-12-29:
BQ3287MT. Tested on both identical motherboards and with BQ3285S / LCS.

No boot (black screen). Some beeps are but didn't check about meaning (it
is MrBios)

necroware, 2022-12-29:
If it is beeping, the RTC module is probably ok. Usually the boards doesn't even get to the POST codes if the RTC module is dead. If it is beeping, then the basic initialization has been done successfully.

adamz464, 2023-01-17:
Accidentally got Alt12887 with DS12885S chip. It worked with this one picky
motherboard (Alaris Cougar).
So I desoldered DS chip and prepared next unit of NWX based on it. Result
the same (no screen).
Must compare pinout of these proejcts because must be difference between
them...

adamz464, 2023-01-17:
Found issue: #21 pin is present in ALT. After soldering this leg back to
your module then mobo boots

adamz464, 2023-01-26:
and...

> Message ID: ***@***.***>
>

adamz464, 2023-02-28:
AND

necroware, 2023-02-28:
Ok, most mainboard don't use that pin, but yours seems to use it. This is the first one so far. I'll have to update the documentation and test what happens if I use it with pin 21 in another boards.