necroware/nwX287 #24: Time saving but setings erasing

486-dx, 2025-07-22:
So my nw287 works it saves my time  but it wont save my hard drive configuration i dont know what to do

486-dx, 2025-07-22:
So the mainbord is genoa ga-486 am the bios is award bios l4854303 the rtc is on the picture the rtc is socketed pins cut of are visible on the picture the batery tests out at 3.24 volts  and  other things seem ok the chip soddered  is ds 12885s  and  i dont think i see any write protection jumpers i didnt  add info because i tought you would not respond 

[image]

[image]

[image]

necroware, 2025-07-22:
> i didnt add info because i tought you would not respond

I'm active in the community and as a software developer since many many years. Please don't get wrong, but the chance for a bad ticket to be ignored is much higher. If you really in search for help, try to create your tickets as good as possible, otherwise there is no point to create those anyway. 

Regarding your issue Genoa ga-486 is not known to me. Do you mean Gigabyte? In such a case it's better to find the board on the retro web project and link it in the issue. You didn't mention which RTC module was originally used in your board, but if you still have it, compare which pins are cut off there. From the picture of your nwX287, I see for example, that pin 21 is still soldered on your module. That pin is responsible for memory reset and if this pins is bound to ground on your board it will erase the internal memory all the time. May be you want to remove that pin. Also your chip looks kind of sketchy tbh. Unfortunately there are a lot of fakes out there.